robotdotnet/WPILib #2: Classes not yet completed

ThadHouse, 2015-05-14:
ADXL345_I2C
ADXL345_SPI
AccumulatorResult
AnalogAccelerometer
AnalogOutput
AnalogPotentiometer
BuiltInAccelerometer
CANJaguar
CANTalon
CameraServer
Compressor
ControllerPower
DoubleSolenoid
GearTooth
Gyro
I2C
PowerDistributionPanel
Preferences
Replay
SPI
SerialPort
Solenoid
SolenoidBase
Ultrasonic
Utility

NetworkTables

Will cross out as completed. There are more, but these are basically what are needed.

ThadHouse, 2015-05-15:
ControllerPower and PowerDistributionPanel added in https://github.com/robotdotnet/robotdotnet-wpilib/commit/6cad27786fca087f8eaf0a6e49f4520538550076

ThadHouse, 2015-05-18:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/a2d063fe10885eda7c1fc68be135fb0f791609d5

ThadHouse, 2015-05-20:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/a491cd0ebb4a9effec0941d283f3a42918eb3bdf Added:
AccumulatorResult
AnalogAccelerometer
AnalogOutput
BuiltInAccelerometer
Gyro
Accelerometer Interface
Relay

jkoritzinsky, 2015-05-21:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/9abf158b950955b21221ed94bd224d77f92c4979 Added:
CANTalon (as much as possible with current HAL).

jkoritzinsky, 2015-05-21:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/72ff752aae6ad95d6fe4abd79ed64c96328e72ac Added:
CANTalon (complete)

jkoritzinsky, 2015-05-22:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/a1baf5d4cf10eba3a8d1d6baad8d5648e29af5b4 Added: I2C

jkoritzinsky, 2015-05-22:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/a5ed3b82d0bedc7f5febfa34f98997bff33cfc9b Added:
SPI

jkoritzinsky, 2015-05-24:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/d828ecf62e08ec24858fcd79b2e37a9d345f46d8 Added:
Ultrasonic

jkoritzinsky, 2015-05-24:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/1a4880b1a277503ce93b03da8e9214d05a6444bb Added:
ADXL345_I2C
ADXL345_SPI

ThadHouse, 2015-05-24:
I'm working on preferences. Need to figure out how Hashtable and Vector will move easiest over to C#

jkoritzinsky, 2015-05-24:
For Hashtable I would suggest Map<K,V> and for Vector I would suggest any implementation of ICollection<T> or IList<T>.

ThadHouse, 2015-05-28:
https://github.com/robotdotnet/robotdotnet-wpilib/commit/582ed91949361447dd0f2749109f203aec9d2f98 Added:
CANJaguar

ThadHouse, 2015-05-28:
As of https://github.com/robotdotnet/robotdotnet-wpilib/commit/4e18d9e60484de4d84647ef8226eaf7593c6c05c:
Preferences
Utility

CameraServer needs all the NI vision stuff. Both Python and Java use a python script to automatically generate bindings for the NI vision libraries. I don't know enough python to modify that script for DotNet. Maybe talk to the original author and see if we can get a version, or an idea on how it works.

ThadHouse, 2015-06-15:
Now that we are down to just a few individual classes working, namely vision, I'm going to close this and create an issue just for that.

closed 2015-06-15 by ThadHouse