robotdotnet/WPILib #12: Figure out NuGet.

ThadHouse, 2015-05-18:
It would be great to use NuGet to distribute the WPILib. It doesnt seem too difficult, but we would need to figure out how to implement it.

jkoritzinsky, 2015-05-19:
How would you feel about creating a MyGet feed for HAL and the WPILib?

ThadHouse, 2015-05-19:
Sure that should work. Looks like an easy way to get everything set up.

ThadHouse, 2015-05-21:
I got a MyGet feed working, and pushed a few builds to it. Is seems to work pretty good. The feed is located here

https://www.myget.org/F/robotdotnet/api/v2

ThadHouse, 2015-06-15:
Done. We have a myget feed for building the code, and an actual nuget feed for hosting the release packages. It was easier just to build release packages and upload them to nuget manually, then to deal with some of the issues in myget. But it still works OK for a build server. Would be nice if it checked pull requests though.

closed 2015-06-15 by ThadHouse