robotdotnet/WPILib #16: HAL P/Invoke wrapper missing needed methods

jkoritzinsky, 2015-05-21:
The HAL generated wrapper is missing some methods that I need for implementing the CANTalon class.  Specifically, all of the PID methods, the Soft Limit methods, and the params_t enum, which alone could be used to implement the rest in .NET and have the .NET code call the general GetParam HAL code.

ThadHouse, 2015-05-21:
Do you know which header file they are located in? I can add it if we can figure out which file the native ones are in.
https://github.com/robotdotnet/allwpilib/tree/master/hal
Here is what is compiled for the HAL.

ThadHouse, 2015-05-21:
OK I just did some looking through the code. The SoftLimits and PID functions are not labeled extern by the HAL, so they cannot be called by P/Invoke. So we will have to use the GetParam and SetParam methods instead. 

For the enum, we can put it in the non Autogeneraged HALCanTalonSRX file, and then create an overload in that file to except the enum, and pass it to the extern function as an int. That way the code exposed to the WPILib is type safe.

jkoritzinsky, 2015-05-21:
Ok. I'll move that in and finish the CANTalon class.

jkoritzinsky, 2015-05-21:
Fixed it up and finished the class.

closed 2015-05-21 by jkoritzinsky