robotdotnet/WPILib #18: Synchronized method vs seperate Lock Object

ThadHouse, 2015-05-22:
Most of the functions in Java that require locks are synchronized functions. I found how to do this in C#, but after reading about it, it looks like it locks on this, which I've always heard it is never a good idea to do this. Should we switch it to using individual locks per class? Would this introduce any race conditions over locking this?

jkoritzinsky, 2015-05-22:
We should use a separate lock object for each instance.  This practice prevents deadlocks by limiting locking to the specified accessors (the current object).  For an example, look at how I implemented the I2C class.

ThadHouse, 2015-05-22:
Ok. Ill go through and change the ones I did. I tried to keep it similar to the java, but I'm guessing they just did that because it was simpler to implement. It might be better to do it right though.

ThadHouse, 2015-05-24:
Still needs to be implemented in the Command classes, but other other ones are changed.

commit edde6aa4 (2015-05-25): Fixes synchronization in commands.  Closes #18
closed 2015-05-25 by jkoritzinsky