robotdotnet/WPILib #19: Some HAL Functions return a status of -63194

ThadHouse, 2015-05-22:
This means NiFpga_Status_VersionMismatch. The functions seem to be returning correct values though. Will have to do some debug with c++ and java to see if they return that as well. But since its a negative number, it should throw in the check status function.

ThadHouse, 2015-06-19:
Looking around, this also happen when trying to initialize a notifier before a digital port has been initialized. I sent an email to WPI about this, but the work around is not super pretty. Looking at the HAL code, it looks like initialize digital port calls a private function, and then returns a struct with a few numbers in it, and doesnt actually create the port in the FPGA. Those are done in later functions. However, there is no function in the HAL to delete this structure without providing a port that has been created in the FPGA. So for now, it looks like we have 3 options.

1. Get the port reference, and ignore it. This will cause a memory leak of 6 bytes. But this would only happen once.
2. Make a version of the struct over on the C# side, and write all the interop code manually to free all the bytes.

Option 1 would be the easiest, but bad programming practice. 
Option 2 could become a problem if the port structure ever changes in the HAL, because normally we don't care about that structure, and we don't even have it mapped in the actual C# code. (The simulator has it mapped, so it would be pretty easy, but could cause later compatibility issues.)

Looking at the HAL digital code, it looks like there is actually no way to free that digital port structure. So whenever one of these digital ports is created, the official WPILib never actually deletes it, even if all references are lost.  So I think I'm gonna implement Option 1 for now, until we have time to go through and modify all the delete functions to do this, because even though its a small memory leak, its still a memory leak...

jkoritzinsky, 2015-06-19:
I agree with your decision for now, but comment the hell out of it so if we or someone else looks at it later they know what is going on.

ThadHouse, 2015-06-19:
oh I plan on that for sure. I just think its bad that even the official HAL has a memory leak, no matter how small it is.

ThadHouse, 2015-06-19:
This is just a reminder to me. I am adding this code to HAL.cs in the HAL-Base project. It is located in the Initialize function.

jkoritzinsky, 2015-06-19:
I agree the official Hal should not have a memory leak and if it does it should be documented for developers of language bindings

ThadHouse, 2015-06-19:
Honestly, I don't think they know. It is something that can only be caused by things only like 0.5% of FRC teams use, and its not big enough for them to notice. Most teams probably are not profiling their memory down at the byte level. Megabyte level maybe, but this would be unnoticeable at that level. Sometime soon I'm going to create an account over at the official repository and submit this and a few more bugs I have found.

ThadHouse, 2015-09-28:
Closing this, and all the other notifier and FPGA bugs, until next season when we have a new FPGA image and can test again. Will create 1 new Issue.

closed 2015-09-28 by ThadHouse