robotdotnet/WPILib #22: Endianness management/conversion

jkoritzinsky, 2015-06-03:
I noticed that some classes (for example CANJaguar) use bit operations to change the endianness of the data from little endian to big endian.  This code is in the WPILibJ (Java version) because Java is always big endian.  We do not have to do any of the bit twiddling unless the roboRIO is big endian because the .NET runtime (and Mono) inherit their endianness from the underlying platform.

ThadHouse, 2015-06-03:
I believe the RoboRIO is little endian. I can double check this though.

ThadHouse, 2015-06-08:
I finally double checked this, and mono on the RoboRIO does run in litte endian. Should we remove those functions? Or wait until we have a chance to test. I don't know when I'll be able to get access to motor controllers to do this.

jkoritzinsky, 2015-06-08:
Remove the endian conversion functions. They'll only create bugs and break the code. Conversely, anywhere that Java casts a byte array to an int without flipping the bits, we need to do that. (C++ is the best guideline here for the bit fiddling.  We should follow C++ on this)

ThadHouse, 2015-06-08:
Well we don't ever have to do the byte array things, because we have IntPtr's and can pass ints by reference to the HAL. The ByteBuffers are the Java workaround for that, and its awful...

jkoritzinsky, 2015-06-08:
I agree, bytebuffers just make life a pain.

ThadHouse, 2015-06-15:
Fixed these in the C# 6.0 branch. Still need to be tested, but I'm going to put class testing in another issue.

closed 2015-06-15 by ThadHouse