robotdotnet/WPILib #23: Updating to C# 6.0

ThadHouse, 2015-06-08:
Should we upgrade to C# 6.0? I would still plan on having it compile into a DotNet 4.5 library, so robot code could be written in VS 2013 if teams would like. But I do like some of the features that C# 6 adds, and we could clean up alot of the code. Plus the instructions would have teams download the newest VS, because I expect 2015 to be released with Win10. I will start work on this in a branch, and we don't have to merge it until we know it works

jkoritzinsky, 2015-06-08:
I'm all for it. If we want to use the features, lets go for it. We can also use this branch for changing the code to use more .NET idioms as well (as we talked about before),

ThadHouse, 2015-06-15:
Fixed in the C# 6.0 branch.

closed 2015-06-15 by ThadHouse