robotdotnet/WPILib #28: HAL static constructor

ThadHouse, 2015-06-17:
I ran into an issue the other day, where the code would work in debug mode, but not release mode. What was happening was that the static initializer for the Driver Station instance was running before the HAL was fully initialized. I know that the python guys were running into this same issue as well, and were just able to add a static constructor in the HAL Base to do this. However since we have 15 different HAL classes, we would have to write 15 static constuctors. Is there any way we can make sure that the HAL is always the first thing initialized without doing this. Otherwise the only way around this would be to have no static instance initializers that involve the HAL.

jkoritzinsky, 2015-06-17:
The underlying issue is that the classes are identified in the CIL with `.beforefieldinit` which means that they can be initialized at any point before use.  There are a few ways to fix this:
* Patch the generated CIL to remove the attribute
Difficult, and in comparison to the other options, not even worth consideration
* Make 15 static constructors with the initialization in there -- This removes the `.beforefieldinit` attribute from the CIL
A reasonable amount of work to move all of the code around.
* Make 15 empty static constructors. -- This also removes the `.beforefieldinit` attribute
Requires the least amount of work.  Definitely should add a comment about this in each of them though.

You can look at the following blog post for a really good explanation: http://csharpindepth.com/Articles/General/Beforefieldinit.aspx

ThadHouse, 2015-06-17:
Actually option 2 is just about as easy as option 3. In the Hal Initialize function, check a flag to see if its already been initialized, and if so just skip it. Then from the 15 status constructors just call the initialize function.

jkoritzinsky, 2015-06-17:
Good point!

ThadHouse, 2015-06-17:
Another thing I found is that looking through that post, it says that the initializer will be called the first time it is referenced, which could happen in a loop. Luckily, since the initalize function calls all the other HAL types, we wouldn't have to worry.

commit 42acb2cf (2015-06-19): Fixed the HAL so it will always be initialized

Closes #28
closed 2015-06-19 by ThadHouse