robotdotnet/WPILib #34: Rewrite Simulator

ThadHouse, 2015-08-03:
Currently a lot of the simulator code was ported from the Python port. However, looking at the native HAL, it seems like some of the implementations are wrong. It would be nice to rewrite some of these so they match the HAL implementation.

ThadHouse, 2015-08-05:
One of the big ones is automated encoder and counter tracking would be very helpful. Its not super hard to do either. The biggest issue is AnalogTriggers, which don't work like DIOs. This is the same reason AnalogTriggers are not supported currently by interrupts. However prelimanary can include only dios, since that was 90% of teams use.

ThadHouse, 2015-09-28:
The work on this is currently being done in the SimulatorRewrite Branch. Should be done this week.

jkoritzinsky, 2015-11-02:
Since this is being covered with more specific issues now, I'm going to close this issue.

closed 2015-11-02 by jkoritzinsky