robotdotnet/WPILib #37: Add a simulator front end

ThadHouse, 2015-08-03:
Python has a WebSim that would be a cool front end to the simulator. It uses WebSockets in a weird way, and I just don't know enough about them to get it working.

In addition, I have been working on a MonoGame simulator base. It looks like that could be really cool. I would have preferred to use Unity, but Unity only supports .NET 3.5.

A 3rd option could be a Windows Form based front end that just shows all of the values and allows them to be modified. That should be fairly easy to implement, expecially with the nested dictionary.

jkoritzinsky, 2015-11-23:
I just ran a little experiment on my local copy and I realized that there's not much that we would have to do to backport our code to Unity.  There were three main pieces that I noticed:

* Caller Info Attributes
  * We can conditionally add in shim versions of the attributes as shown [here](http://www.thomaslevesque.com/2012/06/13/using-c-5-caller-info-attributes-when-targeting-earlier-versions-of-the-net-framework/)
*  Dynamic
  * We'd just need to strongly type or genericify all of the simulator types
*  Monitor.IsEntered
  * We'd just have to find an alternative for this.

ThadHouse, 2015-11-23:
Currently the only dynamic in the simulator is in the Notify method (and the report dictionary, but that could be changed).  The notify method could be just using object. It would be a pain, but it would work.

Monitor.IsEntered isn't supported? Would there even be anything we could use as an alternative to this?

Wouldn't we have to get rid of anything using Tasks as well? Since it only supports .NET 3.5 and I thought tasks was added in .NET 4.0.

ThadHouse, 2015-11-24:
Looks like Tasks are OK. However, if you try and bump down the Extras lib, you would have to change most of AttributedCommandRobot. There is no ReflectionContext, or IReadOnlyDictionary to use. 

In addition, NetworkTables would lose some of the flags functionality, in addition to requiring checking the size of IntPtr to detect 32 vs 64 bit.

I don't think its worth it to backport that far. Hopefully someday Unity ups to Net 4.0 CLR. For now MonoGame works pretty well, and hopefully when the Autodesk simulator releases I can get it ported.

jkoritzinsky, 2015-11-24:
The `ReflectionContext` features of AttributedRobot were just something I thought was cool.  Honestly, I don't expect many (if any at all) teams to use them.  Do we have a frontend made with MonoGame?

ThadHouse, 2015-11-24:
I've been working on one. You have to manually set it up, but its not too hard to do. That's what the simulator patches I've been doing the past few commits are for. I also am working on getting the source code for the new Autodesk simulator, and it doesn't look like that will be hard to implement either.

The hardest part of a simulator is simulating the Driver Station. I'm still trying to think of the best way to do that, and have working joysticks. I was hoping the Sim DS from FIRST would get done, but its been canceled. The Autodesk simulator is just using a loopback adapter, and in all honesty that might be what we end up doing too.

ThadHouse, 2015-11-24:
Currently, the way to properly set up MonoGame is to add the MonoGame project to the solution, Have the MonoGame project depend on the Robot project, keep the robot project as the startup project and then have the robot project start from the MonoGame folder when you click the run button. It actually isnt too bad to setup, and keeps the extension working, keep monogame references correct, and doesnt try and copy MonoGame to the roboRIO on every deploy.

ThadHouse, 2015-12-14:
I've been working on a ton of changes, and actually got creating a MonoGame front end down to just adding a project in the extension. No manual setup required. The extension actually detects the robot project and sets everything correct. I'm still working on the documentation, but for now its much easier.

I also partly got joysticks working. It still needs work, but it should work with both Xbox controllers and regular Joysticks. Need some work polishing the GUI however.