robotdotnet/WPILib #53: Attributed Command Model should add shadow methods

ThadHouse, 2015-10-02:
I'm was working on building a threaded autonomous model, and realized I was going to have trouble if I put all the initialization code into DisabledPeriodic and RobotInit, and then when teams overloaded it, if they forgot to call the base function, the code would not run properly. What I am doing there, and what we should do in AttributedRobot is copy the iterative robot StartCompetition loop, and then add our own shadow methods so they will always get called even if teams override the functions.

jkoritzinsky, 2015-10-02:
The other thing we could do is change the methods that everyone overrides for the Attributed Command Model.  Since we created that model, we can do that.  I'll set that up right now.

ThadHouse, 2015-10-02:
If possible could we get some example code put in the documentation for that? Since its new and not official FRC code we have to be extra good with documentation.

jkoritzinsky, 2015-10-02:
I'll get documentation done for it.  I'll actually assign myself this issue so I don't forget.

jkoritzinsky, 2015-11-02:
I've gotten the documentation done for this, so I'm going to close this issue.

closed 2015-11-02 by jkoritzinsky