robotdotnet/WPILib #55: Todo For 2016 Season Release

ThadHouse, 2015-10-29:
Hey @jkoritzinsky . I'm creating this list to include everything we need to get done before kickoff 2016, or the day of in some cases. I'm thinking we should have 3 priorities.

Priority 1: Must Be Done
Mono Installation Script - Should be done through the extension
Change RoboRIO mDNS name - This got changed in the 2016 image, so can't be tested, but we need to find everywhere this needs to be changed.
Add any new classes added to the WPILib: I think everything currently added has been put in, but there might be more coming in the next few months.
Documentation: We really need to clean up the documentation and the readmes.

Priority 2: Would like to get done
Release Build Script - This can be done manually fairly simply, but an automated tool to do it would be very nice to have.
Get a Driver Station Simulator - We need a simple one for unit testing, but we might need this for team simulator use too. The WPILib simulation library still has not been touched, and I don't know if it's going to get done by the season.

Priority 3: If we finish everything above
Simulation Front End: I started one of these in WinForms a few months ago that would let us control every output and read every input individually. Shouldn't be too hard to finish, so might get pushed up a bit.
Simulation Gaming Front End: An easy to create simulator based off of MonoGame would be cool to have. It's not too difficult to set up manually, but if we could automate this it would be nice.

As we finish stuff we should just cross it off here.

jkoritzinsky, 2015-10-29:
I reformatted this issue as a checklist.  I agree with all of these priorities.  Lets get to work! :+1:

jkoritzinsky, 2015-10-29:
I created a new Sandcastle project for the documentation and I was able to get it working (showing remarks and examples).  I'm going to do a little polishing up on some of our documentation and then I'll commit and upload our updated docs.

ThadHouse, 2015-10-29:
Does it pull the NetworkTables solution from Git before building? The one currently in the WPILib project works for just WPILib, however its does not grab NetworkTables. Or do you know of a way to merge 2 sandcastle builds afterward?

jkoritzinsky, 2015-10-29:
Since we are using two separate repos for WPILib and NetworkTables, it does not currently.  I can make NetworkTablesCore a sub-repo of WPILib so that I can make it grab it.  Or, I can make a separate repo for the documentation generation and make both this repo and the NetworkTablesCore repo subrepos so we can have them all in one repo without polluting each others.

ThadHouse, 2015-10-29:
That was my original goal for https://github.com/robotdotnet/Release-Tools, I just ran out of time to do so. Was thinking we could have the entire build and release process in its own repo. 

https://github.com/robotdotnet/robotdotnet-tools/tree/master/ReleaseBuilder Here had what I was using to build releases. Note that it is absolutely awful...

jkoritzinsky, 2015-10-29:
Ok.  I'll set it up in that repo.

jkoritzinsky, 2015-10-29:
Unified documentation system is set up.  Now all we have to do is finish documenting everything and clean up the default Sandcastle project.

ThadHouse, 2015-11-04:
Mono install script should be done. Needs to be tested with an actual RoboRIO, and I want to refactor the extension a little bit, but it does work.

ThadHouse, 2015-11-04:
MDNS changes are done as well and in the 2016 branch in the extension. Don't want to merge until we have the new image.

ThadHouse, 2015-11-24:
Licensing is done, a simple VB template was added to the template, And I've updated most of the readmes. Will double check the rest of them, then check it off.

ThadHouse, 2015-11-25:
Prebuild script is done for WPILib. NTCore we still need to figure out where those native DLL's will be hosted.

ThadHouse, 2015-11-30:
I got access to the Beta image from FIRST, so I will be merging any changes required for the new image this weekend.

ThadHouse, 2015-12-04:
Got the RoboRIO back and tested the 2016 image. Had to make a few changes, but everything still works well. Also upped the mono version to 4.2.1 from 4.0.1.

ThadHouse, 2016-01-07:
We got everything needed for release done. Still have a few of these things that could be better, but for now we are good.

closed 2016-01-07 by ThadHouse