robotdotnet/WPILib #77: Simulating Gyroscopes

ThadHouse, 2015-12-01:
I'm trying to figure out how to simulate Gyro's, and its looking to be extremely difficult. There are too many dynamic values to deal with, and its kind of a mess. What does everything else think about making the gyro class detect a simulator, and then skipping all the messy calculations and just grabbing raw angles and values from the Sim. Seems like it would be much easier and less prone to errors.

jkoritzinsky, 2015-12-01:
[image]

On a more serious note, let's do the easier way.

ThadHouse, 2015-12-01:
Lol

Ok. Its in the gyro branch, which I'm waiting to merge until the needed changes get made to the HAL upstream. That should be done soon.

And my goal is to do that in as few places as possible. That will currently be the only place where the real code is different from the sim code. Everywhere else it was able to be worked around.

ThadHouse, 2015-12-01:
I did get to use a version of my favorite hack to get it working.

float f = 22.22;
int i = *(int*)&f;

float o = (*float*)&i;

:+1:

closed 2015-12-02 by ThadHouse