robotdotnet/WPILib #79: Combining HAL Dlls

ThadHouse, 2015-12-02:
How would everyone feel about moving the contents of HAL-Simulation and HAL-RoboRIO to HAL-Base, into their own folders in the assembly. I've been running into alot of issues with the way it currently works. If the application is ran from a different directory then the reflection Dll's it has issues. In addition, debugging is a pain, since reflected types do not seem to be entering the debugger correctly. Moving everything to 1 assembly would immensely clean up the the loading code and make debugging easier, and I don't see any downsides to doing so. It would also remove a few hacks we have currently for the circular dependencies needed between the 2.

ThadHouse, 2015-12-02:
One other thing it does is allow teams to easily write custom simulator implementations for things like SPI and I2C devices. Direct access to the simulator code would be much easier for things like that.

jkoritzinsky, 2015-12-02:
I'm all for that. Let's do it.

ThadHouse, 2015-12-02:
Ok. Im working on it in a seperate branch. Also merged the 2 pull requests a little early, because its going to be very difficult to merge anything old after this.

ThadHouse, 2015-12-02:
Done. Make sure to do a fresh pull before working on anything, because things are going to have changed.

closed 2015-12-02 by ThadHouse