robotdotnet/WPILib #99: Entire solution does not build on Linux

ThadHouse, 2016-01-14:
System.Reflection.Context does not seem to exist in Mono, which cause the unit test building to fail.

jkoritzinsky, 2016-01-14:
I'll work on making replacement for CustomReflectionContext (which is the class from that assembly)

jkoritzinsky, 2016-01-14:
After looking at the code, I can't get access to enough of the internals of the reflection stuff to make a replacement. I can either remove the ReflectionContext code or we can make the tests that test it conditionally compile on Linux.  It'll probably be better to just remove the code since CustomReflectionContext is really the only way to use it.

ThadHouse, 2016-01-14:
Ah. Yeah if we can't use it on Mono on the robot there's no point in having it.

closed 2016-03-28 by jkoritzinsky