robotdotnet/WPILib #105: Change documentation system to DocFX

jkoritzinsky, 2016-03-28:
DocFX is simpler than Sandcastle and has official support from Microsoft. Additionally, the search/filter function is Javascript-based, so it will work even when we host on GitHub Pages. It also looks better.

ThadHouse, 2016-03-28:
That looks really cool. Glad to see an officially supported documentation tool from microsoft again.

ThadHouse, 2016-03-28:
It still looks like we're going to need the individual sandcastle projects to generate the xml docs, but that's fairly easy as that does not require maintaining. But it would for sure be nice to get the site completely moved over.

ThadHouse, 2017-01-01:
Got this started over on the https://github.com/robotdotnet/WPILib-Docs repo. Sandcastle didn't support building web docs for project.json, so we had to move. This is much easier however.

closed 2017-01-01 by ThadHouse