robotdotnet/WPILib #112: CANTalon and other removed classes

msoucy, 2017-01-14:
As of 2017, the CANTalon and other classes were removed from the library. The comment for fb5fecc7a037daa7e2da3a837bcc54e73a35238d states that they will be moved to another location, but no replacement location has been given.

ThadHouse, 2017-01-14:
Grab the FRC.WPILib.CTRE package off of Nuget. I'll update the docs tonight. Sorry about that.

ThadHouse, 2017-01-14:
That should be the only removed class. Anything else missing that you need?

msoucy, 2017-01-14:
That appears to be all that we need at the moment. Thank you!

ThadHouse, 2017-01-15:
https://robotdotnet.github.io/tutorials/ctre.html

I added a page to the docs about this. Thanks for reminding me!

closed 2017-01-15 by ThadHouse

msoucy, 2017-01-21:
For history's sake - this was done aed54c5cc7623f7c51fa2299c17c75fb24a23a28