robotdotnet/WPILib #118: Ordered Chooser

msoucy, 2017-02-21:
For feature parity with wpilibsuite/allwpilib#492, the chooser should be ordered.

See wpilibsuite/allwpilib#491 for rationale

ThadHouse, 2017-02-21:
This is going to be a much harder fix in C#, since we don't have an equivalent in the .NET framework. I don't think C++ does either. In addition, I think we were actually looking at reworking sendable choosers for 2018, so we'll see. Most likely that PR won't get merged this season anyway.

msoucy, 2017-02-25:
Slight nitpick - .NET does have an [OrderedDictionary](https://msdn.microsoft.com/en-us/library/system.collections.specialized.ordereddictionary(v=vs.110).aspx), though it does seem outdated. Does this planned rework remove the disparity with the Java WPILib, where sendables in .NET aren't templated?

ThadHouse, 2017-02-25:
Yeah that ordered dictionary isn't even generic. So definitely no to that. They never created a generic one.
TBH I had totally forgotten we made them generic in Java. That's not directly the reason for the rewrite however. We were looking at it to update some bugs that would be hard to remove from the current implementation. This hasn't been decided yet however.