robotdotnet/WPILib #122: Status for 2018

ThadHouse, 2018-01-01:
Hello everyone. 

I'm sorry to have to do this, but I don't think I'm going to have the time to get this project updated for the 2018 season. I will try this week, but between work and school, and other projects I have just been swamped. The only part I know I won't get to is CAN Talon support, as the CTRE library has went through a massive rewrite and I definitely will not have time to rewrite that and get this working.

If someone wants to attempt to pick this up I will happily help get all the build infrastructure tooling set up, but its more work then just rerunning a build.

Sorry. I will try again for 2019 if there is interest.
-Thad

msoucy, 2018-01-02:
I'm interested in trying to get some of the updates in place. What kind of infrastructure is needed? What are the actual tasks that need to be done? I saw a list of changes to the C++ and Java versions, but there's more than just that correct?

ThadHouse, 2018-01-04:
I'll try an upload a new version of the mono runtime tonight or tomorrow. That will be the first thing that will be required. Then mostly just testing. I pushed a few updates to the wpilib repo to make some of the newer things in the hal work. So once I have mono updated, basically the plugin will need to be updated to install the new version of mono and to check for the 2018 image. Once that happens, just testing. Once that happens I can just tag the repos and they automatically publish to nuget.

For ctre, you'll have to wait for their release to happen this weekend, and then basically write all the native entry points and the new wrapper. None of the old code for can talon works anymore.

msoucy, 2018-01-04:
I've been attempting to make some of the user-visible changes for 2018, but it turns out that NetworkTables also needs to be redone... Taking a look at that now, but I'm not sure how much I'll be able to do.

danlucka, 2018-01-07:
How much work will be required to get this going?  Maybe I can help?  I was looking forward to using RobotDotNet this season.

ThadHouse, 2018-01-10:
I got enough done to publish a 2018 extension. So anybody who wants to use this in 2018, it now runs. However, it has not been tested at all with any real code, just enough so that it boots.

 Note it is missing 2 things. First is AutoSPI, and by extension the ADXRS_540 gyro. To fix this, the new HAL calls will need to be added, and then the new high level wrapper copied from java and rewritten.

CANTalon is the other thing missing. Basically, the whole support for that gets to start from scratch. Downloading the native library from CTRE, writing the P/Invoke wrappers, and then writing the wrapper from the new phoenix lib.

Also it is missing any new WPILib functionality for 2018. NetworkTables is still at the 2017 version, so any of the new functionality from that will need a NetworkTables rewrite.

danlucka, 2018-01-12:
Thank you for all your work!  If I can help, please contact me (email is on my profile).

Brzzrs, 2018-04-16:
I know I'm bringing up a dead topic, but I'm interested in this mainly for the NetworkTables, any hints on the difficulty of updating the ntcore implementation to work for 2018? I'm open to all help, suggestions, and anything in general, thanks!

Brzzrs, 2018-05-01:
> It would need a complete rewrite. From the interface to the native library
> to the user facing code itself. I had started on it last summer, so I have
> a bit of the user facing side, but not much else.

Would you consider working with me to update it? this would allow huge
advantages for our driver-station as the java framework for user interfaces
is absolute trash, the idea of using actual c# push buttons and such would
highly benefit my drive team. Let me know, thanks! (all I need from the
library is to get NetworkTables running as a client on my driver-station PC)

ThadHouse, 2018-06-01:
@Brzzrs Can you create an issue in the NetworkTables repo? I'm actually working on getting the wrappers around NT4 implemented in the `newnt4` branch. Nothings been pushed, but should be soon

jrmoreno1, 2022-10-22:
Current status?